riscv-non-isa/riscv-iopmp #4: Equations for triggering an interrupt is not clear when an IOPMP instance hits no entry or multiple non-priority entries

tyshyu, 2024-06-07:
Hi all,

The latest commits (3369da4 7a76c5d) add some equations for describing how to trigger an interrupt in Section 2.7 (chapter2.adoc):
```
interrupt_on_read = ERR_CFG.ie && !ENTRY_CFG.sire
interrupt_on_write = ERR_CFG.ie && !ENTRY_CFG.siwe
interrupt_on_execution = ERR_CFG.ie && !ENTRY_CFG.sixe
```
The aforementioned equations are not clear if an IOPMP instance has no hit and multiple hits on non-priority entries.

For a read violation, would it be like the following equation?
```
interrupt_on_read =
ERR_CFG.ie && (no_hit || 
  ! (
          (entry_hit[0] && ENTRY_CFG(0).sire)
      ||  (entry_hit[1] && ENTRY_CFG(1).sire)
      ||  (entry_hit[2] && ENTRY_CFG(2).sire)
      ...
      ||  (entry_hit[N-1] && ENTRY_CFG(N-1).sire)
    )
)
```
`no_hit` means an IOPMP instance hits no entry.
`entry_hit[i]` means an IOPMP instance hits or partially hits entry i.
`N` means implemented IOPMP entries.

The equation means any entry is hit and its `sire` bit is 1, the interrupt will be suppressed.

channingt, 2024-07-02:
For non-prioritized entries:
1. if an entry is hit, then no interrupt generated;
2. if no entry is hit, then generate interrupt.

My understanding is, partial hit is considered the entry no hit. @paul-andes , what's your thought? We can discuss on the meeting.

paul-andes, 2024-07-09:
The discussion will keep on in the following link:
https://lists.riscv.org/g/tech-iopmp/message/293