riscv-non-isa/riscv-iopmp #7: IOMMU and IOPMP intergreation method

romanheros, 2024-06-17:
Throughout the IOPMP specification, it doesn't have description its relationship with IOMMU. I want to know how to intergrete them togather. I think at least we should point out the order of them to check DMA address. As IOMMU will really translate IOVA to HPA, it is also need explicitly point out what is IOPMP check, the IOVA or HPA？

tyshyu, 2024-06-17:
The following comment is my thoughts. Please correct me if something is wrong or other use cases are not applicable for it.

Hypervisor (HS-mode software) or software which has lower privilege than hypervisor manages IOMMU.
Secure monitor (M-mode software) manages IOPMP and only the secure monitor is trusted in the system.
The IOPMP could check every SPA (supervisor physical address) access from the IOMMU, including page table walk and in-memory queue interface access. The term "SPA" would be equivalent to HPA if HPA represents host physical address.

paul-andes, 2024-07-02:
In an SoC having both IOMMU and IOPMP, the IOPMP is expected to protect the data belonging to the secure monitor (running in the M-mode of the secure core) and isolate between the hypervisor and trusted execution environment(s) for more complicated cases. Thus, the transactions issued from IOMMU, regardless of the page walker or data access, should be confined within some specific areas defined by a secure monitor. The IOPMP can be either source-enforcement or destination-enforcement. In the software's view, the IOMMU is operated by OSes and/or the Hypervisor, while the IOPMP is the tool of the secure monitor.

We will add some descriptions to the specification.

closed 2024-12-19 by paul-andes