riscv-non-isa/riscv-iopmp #10: Should IOPMP support EPMP like pmp？

leBsky, 2024-10-09:
EPMP is an extension for PMP. Should IOPMP support EPMP?

tyshyu, 2024-10-11:
Does it mean Supervisor Memory Access Prevention (SMAP) and Supervisor Memory Execution Prevention (SMEP) feature of ePMP? IOPMP covers the feature. Please feel free to share which ePMP features should be supported in IOPMP.

tyshyu, 2024-10-14:
I think IOPMP is not required to support the SMEPMP extension because IOPMP can check machine mode access in current spec.

AFAIK, the SMEPMP extension is to check machine mode access by different behavior of pmpcfg.l/r/w/x. Alternatively, IOPMP can check machine mode access by an RRID which is from an RISC-V core or cluster. For example, RRID 0 indicates machine mode access from a RISC-V core and RRID 1 indicates non-machine mode access from the RISC-V core, and Memory Domain 0/1/2 indicate machine mode access exclusive regions/non-machine mode access exclusive regions/shared regions. If a non-RV device (DMA controller, GPU, or AI accelerator) has some states like machine mode and non-machine mode of RISC-V, IOPMP can do similar way for non-RV devices.

However, the RRID assignment is implementation-dependent, checking machine mode access may not be supported by chip vendors' decision or limitation of devices/RISC-V cores. The devices/RISC-V cores may not output machine mode access indication or non-machine mode access indication for the RRIDs when they issue a request for data or instruction to devices or memories.