riscv-non-isa/riscv-iopmp #9: IOPMP/SPS behavior clarification

leBsky, 2024-10-09:
When IOPMP/SPS is implemented, is is said in spec  "IOPMP/SPS shall check read and write permission on both the SRCMD table
and entries, a transaction fail the IOPMP/SPS check if it violates either of the permission settings"

Since SRCMD table is looked up first then the permission from SRCMD_R/W can be checked firtly. If the permission from SRCMD_R/W are all 0, then for a read or write request, does it still need to check the IOPMP entries because it has already get a denied permission ?

tyshyu, 2024-10-11:
From my perspective, it doesn't need to check the IOPMP entries, because the behavior of the IOPMP is not changed if the IOPMP additionally checks IOPMP entries. As for error record in IOPMP, I think the error record should be record when the first error is happened and settings for error record is enable.  Therefore, the error record is also not changed if the IOPMP additionally checks IOPMP entries.

leBsky, 2024-10-11:
> From my perspective, it doesn't need to check the IOPMP entries, because the behavior of the IOPMP is not changed if the IOPMP additionally checks IOPMP entries. As for error record in IOPMP, I [...]

Assume it is not required to check IOPMP entries in this case, then the interrupt or error response or error record behavior will only be controlled by ERR_CFG register. The Entry_CFG should have no influence on the behavior of interrupt, error response and error record.

tyshyu, 2024-10-14:
> Assume it is not required to check IOPMP entries in this case, then the interrupt or error response or error record behavior will only be controlled by ERR_CFG register. The Entry_CFG should have no [...]

My thoughts are the same as yours.

However, the current IOPMP spec does not seem to explicitly clarify the behaviors. Only "When SPS extension is enabled, the IOPMP checks both the R/W/X and the ENTRY_CFG.r/w/x permission and follows a fail-first rule." in Chapter 5 reflects the rules we mentioned. It will be better if the behaviors will be clarified in IOPMP spec.

tyshyu, 2024-10-16:
From the previous comments, interrupt, bus error response and error record behavior of SRCMD_R/W only be controlled by ERR_CFG register. It can not be controlled locally. I am not sure if it should be controlled only by ERR_CFG register, or there are additional options.

I have opened discussion about this in IOPMP TG mailing list in the following link:
https://lists.riscv.org/g/tech-iopmp/message/320

tyshyu, 2024-10-23:
My previous comments are not accurate.

From the message in IOPMP TG mailing list (https://lists.riscv.org/g/tech-iopmp/message/323). IOPMP should check permission on both the SRCMD table and entries.

leBsky, 2024-10-23:
> My previous comments are not accurate.
> 
> From the message in IOPMP TG mailing list (https://lists.riscv.org/g/tech-iopmp/message/323). IOPMP should check permission on both the SRCMD table and entries.

Let's consider the following case
1. SRCMD table check fail, IOPMP entry full hit, check pass, then the interrupt/bus error/error record behavior are controlled only by ERR_CFG
2. SRCMD  table check fail, IOPMP entry full hit, check fail, then the interrupt/bus error/error record behavior are controlled by both ERR_CFG and local IOPMP entry error control bits.

Do I understand correctly?

tyshyu, 2024-10-30:
According to the messages in IOPMP TG mailing list, I have the same understanding as you.

leBsky, 2024-11-25:
> According to the messages in IOPMP TG mailing list, I have the same understanding as you.

[image]
See the formula here, If SPS check fails and IOPMP entry check passes with suppressing bits of all the matched IOPMP entries set, then the interrupt will still be suppressed, Which is different from what talked before.  Before we thought when SPS check fails and IOPMP entry check passes, then the interrupt/bus error/error record behavior are controlled only by ERR_CFG but the formula in Spec indicates that ENTRY_CFG should still be considered in this case.

tyshyu, 2024-12-05:
Thank you for your feedback. Error reactions about SPS and `SRCMD_PERM(m)` actually are needed to be described in Section 2.7.

However, our previous understanding is incorrect. The following is the reason.

According to figure 2 in spec V0.9.2-RC2, permissions in SRCMD table are transferred to IOPMP entries and are checked in IOPMP entries. An entry has two types of permission to be checked: (1) per-entry permission (`ENTRY_CFG(i).r/w/x`), and (2) permission from SRCMD table. Therefore, interrupt/bus error/error record behavior are controlled by `ERR_CFG` and local IOPMP entry error control bits when any permission (type (1) and type (2)) in an entry check fail, because the check is failed in a specific entry.

The following case you mentioned:
> 1. SRCMD table check fail, IOPMP entry full hit, check pass, then the interrupt/bus error/error record behavior are controlled only by `ERR_CFG`

It should be:
> An IOPMP entry full hit, corresponding permission bit of SRCMD table check fail, permission bit of the entry check pass, then the interrupt/bus error/error record behavior are controlled by `ERR_CFG` and local IOPMP entry error control bits.

The behaviors will be clarified in spec V0.9.2-RC3 (next release).

leBsky, 2024-12-06:
> Thank you for your feedback. Error reactions about SPS and `SRCMD_PERM(m)` actually are needed to be described in Section 2.7.
>
> However, our previous understanding is incorrect. The following is the [...]

Can I get the conclusion like this " As long as there is an IOPMP entry hits, then the interrupt/bus error/error record behavior should be controlled by `ERR_CFG` and local IOPMP entry error control bits together. And only when no IOPMP entry hits, then the interrupt/bus error/error record behavior are controlled by `ERR_CFG` alone.

tyshyu, 2024-12-11:
Yes, you can get the conclusion but it is only for priority entries. The conclusion is not accurate for non-priority entries. "A non-priority entry matches all bytes of a transaction and the permission check is success" is hit for non-priority entries.

The more accurate description for non-priority entries should be like this "As long as there are IOPMP entries matches all bytes of a transaction and all entries have permission violation, then the interrupt/bus error/error record behavior should be controlled by `ERR_CFG` and local IOPMP entry error control bits from all matched entries together. And only when no IOPMP entry matches all bytes of a transaction, then the interrupt/bus error/error record behavior are controlled by `ERR_CFG` alone"

closed 2024-12-19 by paul-andes

tyshyu, 2025-01-06:
> Can I get the conclusion like this " As long as there is an IOPMP entry hits, then the interrupt/bus error/error record behavior should be controlled by `ERR_CFG` and local IOPMP entry error control [...]

Yes,  you can get the conclusion. My previous comment is obsolete.
Priority entries and non-priority entries have same definition about hit.  V0.9.2-RC3 has clarified this in Section 2.6, 2.7, and table 2.