riscv-non-isa/riscv-iopmp #11: Does IOPMP support MSI?

zhuotianshu, 2024-10-11:
Current Spec didn't explicitly specify the delivery mode of IOPMP interrupts. But as I understand it, if it support MSI, there should be some registers to hold the target and IID of the interrputs, just like RISC-V IOMMU.

tyshyu, 2024-10-11:
IOPMP does not support MSI in current spec. Is there any demand for MSI support in IOPMP?

zhuotianshu, 2024-10-11:
Actually I'm not sure. But if an IOPMP supports MSI, it can be easily integrated with an IOMMU that only supports MSI. But it's not a must. Thanks for replying.

chenwhJ, 2024-10-20:
> IOPMP does not support MSI in current spec. Is there any demand for MSI support in IOPMP?

In IOPMP sepc section 2.7, it says that "When an IOPMP detects an illegal transaction, it could initiate three of the following actions. …… Secondly, it could trigger an interrupt.“

tyshyu, 2024-10-30:
In current spec, IOPMP appears to only support wired interrupt so IOPMP haven't the ability to trigger an MSI interrupt. Alternatively, to trigger a MSI interrupt from IOPMP, IOPMP interrupt can be a source of an APLIC (advanced PLIC in RISC-V Advanced Interrupt Architecture). 

> Or else, should IOPMP transfer the MSI to IOMMU to send the interrupt, if an IOPMP is always integrated with an IOMMU?

It should't. As I see it, it is an implementation choice that is decided by SoC designers or IP designers.

tyshyu, 2024-11-13:
@zhuotianshu  @chenwhJ 
From the last IOPMP TG meeting (https://github.com/riscv-admin/iopmp/blob/main/meeting-minutes/2024-1107-IOPMP-meeting-minutes.pdf), IOPMP will support MSI.

gagachang, 2024-11-25:
@zhuotianshu @chenwhJ 
For MSI support, please check latest released spec https://github.com/riscv-non-isa/iopmp-spec/releases/tag/v0.9.2-RC2
We would like to close this issue if you guys do not have more problems after reading.

closed 2024-12-19 by paul-andes