riscv-non-isa/riscv-iopmp #17: Question about Multi-Faults Record (MFR) Extension

YazanHussnain-10x, 2024-11-05:
In the Multi-Faults Record Extension, it states:

> When **ERR_MFR** is read, **svi** sequentially scans all windows from its starting position until a violation is found. Once **svi** overflows, it wraps back to zero.

My question is: if **svi** can wrap around and we need to search all windows, why do we need the record window index (**svi**) in **ERR_MFR**, and why must the scan start from **svi**? If **svi** is specified why we are not only checking the respective record window?

tyshyu, 2024-11-13:
IOPMP automatically scans for the next violated RRID and indexes the next violated RRIDs by `ERR_MFR.svi` when software reads `ERR_MFR`. In other words, IOPMP skips record windows which have no violation. Therefore, software need `ERR_MFR.svi` to indicate which record window is in `ERR_MFR.svw`.

YazanHussnain-10x, 2024-11-13:
Thank you @tyshyu

closed 2024-11-13 by YazanHussnain-10x