riscv-non-isa/riscv-iopmp #19: Understanding IOPMP Models

YazanHussnain-10x, 2024-11-12:
I am currently reviewing the various IOPMP models outlined in the specification and wanted to confirm my understanding. I have created a table that categorizes each model according to the values of `SRCMD_FMT` and `MDCFG_FMT`. Kindly take a quick look and let me know if my interpretation is correct. @tyshyu @paul-andes 

| Model Name       | SRCMD_FMT | MDCFG_FMT | Description |
|------------------|------------|------------|-------------|
| **Full Model**   | 0          | 0          | Max Supported RRIDs: 65536 <br> Max Supported MDs: 63 <br> Uses RRID to obtain SRCMD_EN(H), indicating the associated MDs. Traverse MDs, find associated IOPMP entries, perform address matching, and check permissions. |
| **Rapid-k Model**| 0          | 1          | Max Supported RRIDs: 65536 <br> Max Supported MDs: 63 <br> Uses RRID to obtain SRCMD_EN(H). No physical MDCFG table; each MD has k IOPMP entries: <br> MD0 → 0 to (k - 1) <br> MD1 → k to (2k - 1) <br> MD2 → 2k to (3k - 1), and so on. <br> Traverse IOPMP entries associated with MDs linked to RRID, perform address matching and permission check. |
| **Dynamic-k Model** | 0       | 2          | Same as Rapid-k Model, but k is programmable. |
|   | 1          | 0          | Max Supported RRIDs: 63 <br> Max Supported MDs: 63 <br> No physical SRCMD table; RRID i directly maps to MD i. <br> Associated IOPMP entries: MD(i-1).t to entries < MD(i).t. <br> Perform address matching and permission check. |
| **Compact-k Model** | 1      | 1          | Max Supported RRIDs: 63 <br> Max Supported MDs: 63 <br> No SRCMD table; RRID i directly maps to MD i. No physical MDCFG table; each MD has k associated IOPMP entries <br> Associated Entries: (i × k) to ((i+1) × k - 1). <br> Perform address matching and permission check. |
|  | 1          | 2          | Same as Compact-k Model, but k is programmable. |
|   | 2          | 0          | Max Supported RRIDs: 32 <br> Max Supported MDs: 63 <br> SRCMD_EN(H) and SPS Extension registers replaced with SRCMD_PERM(H). Traverse all MDs, locate associated IOPMP entries, perform address matching, and check permissions. |
|   | 2          | 1          | Max Supported RRIDs: 32 <br> Max Supported MDs: 63 <br> SRCMD_EN(H) and SPS Extension registers replaced with SRCMD_PERM(H). No MDCFG Table; each MD has k associated IOPMP entries. SRCMD_PERM(H) defines permissions but not associated MDs therefore, all IOPMP entries must be traversed. <br> Perform address matching and apply permissions from SRCMD_PERM(H). |
|   | 2          | 2          | Same as above, but k is programmable. |

gagachang, 2024-11-13:
Hi @YazanHussnain-10x, your interpretation correct.
Note that SRCMD_FMT=1 and MDCFG_FMT=0 has model name called "Isolation".
It's unintentionally removed in v0.9.2. It will be added back in next verion.

YazanHussnain-10x, 2024-11-13:
Thank you @gagachang

tyshyu, 2024-11-13:
> Hi @YazanHussnain-10x, your interpretation is correct. Note that SRCMD_FMT=1 and MDCFG_FMT=0 has model name called "Isolation". It's unintentionally removed in v0.9.2. It will be added back in next version.

BTW, Isolation model had been added back in commit 148660ebc9313671bc571bbaa6b3263b6c92a7e8

gagachang, 2024-11-13:
You're welcome @YazanHussnain-10x 
Do you have more questions about IOPMP models ?

YazanHussnain-10x, 2024-11-13:
Not at this time. Thank you for the clarification, @gagachang and @tyshyu.

closed 2024-11-13 by YazanHussnain-10x