riscv-non-isa/riscv-iopmp #36: How to define the value of "specver" in register VERSION

bqiu728, 2024-12-30:
Hi,

Several versions have been released, including v0.9.0, v0.9.1, v0.9.2-RC1, and v0.9.2-RC2.  How should these versions be represented in the `VERSION.specver` register?

Thanks

tyshyu, 2024-12-31:
AFAIK, `VERSION.specver` is not defined currently. It will be defined in ratified version.

bqiu728, 2025-01-02:
> AFAIK, `VERSION.specver` is not defined currently. It will be defined in ratified version.

Thanks.

I see that the current version is in freeze.  Do you have any information that when will it be defined？

Thanks

tyshyu, 2025-01-03:
> > AFAIK, `VERSION.specver` is not defined currently. It will be defined in ratified version.
> 
> Thanks.
> 
> I see that the current version is in freeze. Do you have any information that when will it be defined？
> 
> Thanks

 I have no information about the schedule. At least it will not be defined in v0.9.2-RC3, the next release.

closed 2025-01-06 by paul-andes