riscv-non-isa/riscv-iopmp #39: Clarification in Error Reactions

gullahmed1, 2025-01-10:
section 1:

> The error capture record maintains the specifics of the first illegal access detected, except if the
> following two conditions are held: (1) any interrupt-suppress bit regarding the access is set, and (2) no
> bus error is returned.

section 2:

> If a transaction violates IOPMP rules on multiple non-priority entries which don’t suppress interrupt or
> bus error response, the ERR_REQID.eid should be one of the entries.

The error capture record maintains the specifics of the first illegal access detected, as stated in Section 1, regardless of whether the access corresponds to priority or non-priority entries. For non-priority entries, this record captures the specifics of the first illegal access detected, which is clear. However, Section 2 states that the ERR_REQID.eid should be one of the non-priority entries that violated the IOPMP rules and do not suppress interrupts or bus errors.

This creates ambiguity: while the error capture record holds details of the first illegal access, the ERR_REQID.eid might not correspond to the first entry but to any of the violating entries. Shouldn’t the captured ERR_REQID.eid also belong to the first illegal access entry, for consistency?

gullahmed1, 2025-01-13:
Thank you for your clarification but let's assume a scenario where a transaction needs read and execute permissions. This transaction matches two non-priority entries let's say 10 and 11. Entry-10 gives an illegal read access error and Entry-11 gives an illegal instruction fetch error.

According to section#1, illegal read permission error will be captured in the error capture register.
According to section#2, `ERR_REQID.eid` could contain 10 or 11. 

Now If someone prefers to keep 11 in `ERR_REQID.eid`. Correct me if I am mistaken but won't this bring inconsistency that the error capture record contains an illegal read access error and  `ERR_REQID.eid` says it's because of entry 11 but it's actually due to entry 10?

tyshyu, 2025-01-17:
Thank you for your scenario. 

The error type is identical since it depends on transaction, not entries.

As to "a transaction needs read and execute permissions". I guess it means the transaction is a read transaction, and is also a instruction fetch transaction.  A transaction can't be a read access transaction and a instruction fetch transaction for their meaning. Therefore, the scenario won't be happened.

Let's assume another scenario like the scenario, a transaction needs read and write permission, it can stand for an atomic transaction like atomic swap. In the scenario, one or more entries should give error type "0x02 illegal write/AMO access" if they don't grant their permissions.

However, error type 0x02 in V0.9.2-RC3 is missing some words (currently shows 0x02 illegal write access). It should be "0x02 illegal write/AMO access" like `mcause` "0x7 Store/AMO access fault" in a RISC-V hart, so I will send a PR for the clarification.

gullahmed1, 2025-01-17:
Thank you for the clarification.

closed 2025-01-17 by gullahmed1