riscv-non-isa/riscv-iopmp #42: Typos in Error Reaction - Section 2.7

YazanHussnain-10x, 2025-02-03:
In Section 2.7 - Error Reaction, the following paragraph contains incorrect statements, highlighted below:

The error response is controlled by the global error suppression configuration bit **`ERR_CFG.rs`**, not by the interrupt enable bit **`ERR_CFG.ie`**.

[image]

Similarly in the image below, Errors are not suppressed when **`!ERR_CFG.rs`**, not when interrupt enable is set **`ERR_CFG.ie`**.

[image]

tyshyu, 2025-02-04:
Thank you for your feedback. We will fix it.

tyshyu, 2025-02-10:
Typos have been fixed in new version - Version 0.7.
https://github.com/riscv-non-isa/iopmp-spec/releases/tag/v0.7

closed 2025-02-10 by YazanHussnain-10x