riscv-non-isa/riscv-iopmp #50: IOPMP Configuration Query (No Physical MDCFG Table)

gullahmed1, 2025-03-18:
#### Context:
- **No physical MDCFG table** is implemented.
- Each memory domain (MD) has **k entries**, where `k = HWCFG0.md_entry_num + 1 = 63` (given `HWCFG0.md_entry_num = 62`).
- System parameters:
  - `HWCFG1.entry_num = 128` (total IOPMP entries).
  - `HWCFG0.md_num = 63` (max MD index = 63).

#### Observed Entry Allocation:
- **MD[0]**: Entries `0–61` (62 entries).
- **MD[1]**: Entries `62–123` (62 entries).
- **MD[2]**: Entries `124–127` (4 entries).

#### Questions:
1. **MD[2]** has less than k entries. Is this allowed or should **MD[2]** be made empty?  

2. **Higher MDs (MD[3]–MD[127])**:  
   Are these MDs **empty** (no associated entries) due to the total entry limit (`128`)?

tyshyu, 2025-03-26:
Thank you for your feedback. Spec V0.7 actually has no information for such configurations. We will update related information.

Index range of a MD can >= `HWCFG1.entry_num` but any entry with index >= `HWCFG1.entry_num` in a MD is not available, even if an implementation allows `MDCFG(m).t` >= `HWCFG1.entry_num`. It means:
1. Address mode of the entry is OFF when the IOPMP checks permission of a transaction.
2. The entry is unimplemented when registers of the entry are accessed from control port.

Note: The behavior of unimplemented register is implementation-dependent, which is described on beginning of Chapter 5.

**Answers:**
1. Less than k entries is allowed. MD2 shouldn't be made empty.
2. These MDs are empty.

gullahmed1, 2025-04-03:
Thank You for your response.

closed 2025-04-03 by gullahmed1