riscv-non-isa/riscv-iopmp #54: Array entries belonging to more than one Memory Domain

just-for-fun-too, 2025-04-09:
In Section 2.5, the following is stated regarding entries and Memory Domains:

> “Each entry in the array can belong to at most one memory domain, while a memory domain could have multiple entries.”

I presume this has to be checked at execution time once all the MDs associated with an RRID have been determined and the MDCFG table is indexed to get the entry numbers associated with each MD. 

If it is found that an entry is associated with more than one MD, which error is signaled? 

Do we make the entry list for that RRID empty and signal error 0x5 indicating no rule has been hit?

tyshyu, 2025-04-10:
> If it is found that an entry is associated with more than one MD, which error is signaled?

MDCFG Table constrains that an IOPMP entry must belong to at most one memory domain. Therefore, no error is signaled.

> Do we make the entry list for that RRID empty and signal error 0x5 indicating no rule has been hit?

Yes. An IOPMP signals error 0x5 when the incoming transaction carries a valid RRID and no entry is hit for the RRID.

just-for-fun-too, 2025-04-10:
@tyshyu Thanks for the answer. However it is still not clear what the specification expects an implementation to do. 

You say this is a constraint. Okay.  But what if the software makes a mistake and doesn't follow the constraint - buggy software, happens once in a while. So then:

1. Do you expect the implementation to check this at execution time (relatively easy) or at programming time (not so easy) to see if the constraint is being adhered to or not?
2. If it is not (meaning an entry belongs to more than one MD) what do you expect the implementation to do? Should it:

      a-  Generate an error? IF it's an error which error?
      b-  Ignore the fact that that an entry belongs to multiple MDs?
      c- Do something else?

The question is if Software didn't follow the constraint what do you expect the implementation to do?

Thanks!

tyshyu, 2025-04-11:
> The question is if Software didn't follow the constraint what do you expect the implementation to do?

The constraint is a hardware implementation constraint of MDCFG Table in Section 3.3.1, not a software implementation constraint. 

> 1. Do you expect the implementation to check this at execution time (relatively easy) or at programming time (not so easy) to see if the constraint is being adhered to or not?`

An IOPMP instance should keep "an IOPMP entry must belong to at most one memory domain" based on MDCFG Table in Section 3.3.1.
So it is at execution time if I understand "execution time" correctly.

The remaining question is that descriptions in Section 3.3.1 needs clarification. That's discuss it in other issues.

closed 2025-06-13 by paul-andes