riscv-non-isa/riscv-iopmp #56: ENTRY_CFG(i).a issue

just-for-fun-too, 2025-04-09:
In Section 2.5 it states:

> “… software should (1) avoid adopting TOR for the first entry of a memory domain, or (2) set an OFF for the last entry of a memory domain with the maximal address during programming the IOPMP.”

If these conditions are not followed by software, should an error be signaled?  If so, which error? Or should this be ignored, as it seems to be implied in the specification?

luisccc, 2025-04-10:
I do not know if my assumption is correct, but I will give my two cents. 
I think that note is just a warning for people programming the IOPMP, as it is an edge case which arises from the interaction between MDs and entries. It should not give an error as the IOPMP is working correctly (albeit not as intended), the configuration is just not correct.

tyshyu, 2025-04-11:
An error shouldn't be signaled.
Current IOPMP specification only gives the programming notice for the edge case.
First entry of a MD with TOR still refers the last entry of previous MD.

closed 2025-04-11 by just-for-fun-too