riscv-non-isa/riscv-iopmp #57: Improperly programming an IOPMP

just-for-fun-too, 2025-04-09:
In Section 3.3.1 it states

> “The IOPMP behavior of improperly programming them is implementation-dependent as long as 

1. an entry must belong to at most one memory domain and 
2. the index of an entry in a lower-indexed memory domain should be lower than that in a higher-indexed memory domain.”

So (1) and (2) identify the two ways in which the MDCFG Table Format 0 can be improperly programmed. The question is what, if any, are the additional ways in which the table can be incorrectly programmed.

If these are the only ways in which it can be incorrectly programmed, what, if any, are the error numbers that should be generated?

tyshyu, 2025-04-10:
There are no additional way.
(1) and (2) are constraints of MDCFG Table. When MDCFG Table is improperly programmed, MDCFG Table behavior is implementation-dependent but it should follow the constraints.

IOPMP doesn't generate any configuration error in current IOPMP specification.

just-for-fun-too, 2025-04-10:
So if we want to we can generate a "user defined error" (Table 2 Section 2.7). Is that correct?

tyshyu, 2025-04-11:
Correct.

However, the field behavior of MDCFG Table is WARL so the table is not expected to generate any configuration error.

I will mention the MDCFG Table use cases to other issues you have posted.