riscv-non-isa/riscv-iopmp #59: Clarification on MDCFG and MDs

YazanHussnain-10x, 2025-04-10:
#### Context:  
In **Section 3.3.1**, the specification states:  
> MDCFG(m+1).t should be programmed larger than or equal to MDCFG(m).t.

This raises the question: **Does this imply that MDCFG(m).t should initially be set to the maximum supported IOPMP entries?**

In **Section 5.5**, it is mentioned that the default value of the `t` field is either **implementation-dependent** or **Don't Care**. Why is the default value considered "Don't Care"?

#### Questions and Observations:  

1. **Behavior at Reset**:  
   Assuming the MDCFG table is initialized to zero at reset:  
   - If software programs MD2 and associates it with an RRID while MD3 remains unassociated, the condition in Section 3.3.1 (i.e., `MDCFG(m+1).t >= MDCFG(m).t`) is not satisfied.  
   - Is this scenario allowed?

   My understanding is that as long as an MD is not associated with any RRID, the condition in Section 3.3.1 does not apply. Is this interpretation correct?

2. **Clarification on Restrictions**:  
   The specification imposes the following restrictions:  
   > (1) An entry must belong to at most one memory domain.  
   > (2) The index of an entry in a lower-indexed memory domain should be lower than that in a higher-indexed memory domain.

   - Are these restrictions applicable **only** to memory domains associated with an RRID?  
   - If these restrictions apply to both associated and unassociated MDs, then it seems the software must program the MDs sequentially (e.g., MD0 → MD1 → MD2, etc.). This would prevent configuring MDs in any arbitrary order. Is this the intended behavior?

   If the default value of `MDCFG(m).t` is set to the maximum supported IOPMP entries, associating or not associating an MD would not pose an issue. However, this could compromise security if `MDCFG` is not programmed before `SRCMD_EN`.

#### Example Scenario:  

Consider the following example where, at reset, `MDCFG(m).t = 0` for all `0 <= m <= 62`. Software programs the MDs as follows:  

- `MDCFG(2).t = 8` (Associated with RRID A).  
- `MDCFG(20).t = 64` (Not associated with any RRID).  
- `MDCFG(21).t = 78` (Associated with RRID B).  

This configuration appears valid as it adheres to all stated restrictions:  
- Entries associated with MD2 are: `0 <= entries < 8`.  
- Entries associated with MD21 are: `64 <= entries < 78`.  

However, consider a buggy scenario:  
- The software associates MD20 with an RRID (via a single write to `SRCMD_EH`) without configuring MD19.  
- Alternatively, if MD19 is configured as `MDCFG(19).t = 6`, the conditions still result in a violation where a single entry is associated with multiple memory domains.  

This happens because restriction (2) does not says that the entry index in a higher-indexed memory domain is greater than **all** indexes in lower-indexed memory domains.  

#### Concern:  
This highlights a potential issue:  
- Detecting such misconfigurations during programming is complex, as it requires searching all MDs for duplicate entries.  
- Multiple MDs can be associated with RRIDs in a single `SRCMD_EN` write, further complicating validation.  

I support the solution proposed in **Issue #54** , which suggests runtime detection and error reporting (e.g., **IOPMP Mis-Configured**) in such cases.  

#### Additional Concern:  
If a runtime error is generated due to IOPMP misconfiguration, should it always be reported? If so, it must not be suppressed or discarded. I am uncertain about this point.

#### Proposal for Software Guidelines:  
Shouldn't the IOPMP specification include a **Software Guidelines** section that explicitly outlines the correct programming sequence for IOPMP configuration? For example:  
- When associating a new MD, `MDCFG` should be programmed **before** associating the MD in `SRCMD_EN`. Although this behavior may seem obvious, explicitly documenting it would help prevent ambiguities

tyshyu, 2025-04-11:
Thank @YazanHussnain-10x and @just-for-fun-too for the feedback about MDCFG Table.

I will clarify Issue #54, #55, #57 and #59 in the issue because I think they focus the same issue. It could take a while.

Hi @just-for-fun-too, you could refer the issue if my comments in your issues (Issue #54, #55, #57) are not clear.

tyshyu, 2025-05-20:
Programmers should keep proper MDCFG Table with specific programming order for software portability.
The specification (v0.6) defines the implementation-dependent behavior with minimum restrictions if some programmers accidentally program improper setting.

> #### Context:
> In **Section 3.3.1**, the specification states:
> 
> > MDCFG(m+1).t should be programmed larger than or equal to MDCFG(m).t.
> 
> This raises the question: **Does this imply that MDCFG(m).t should initially be set to the maximum supported IOPMP entries?**

It doesn't. They can be any value.

> 
> In **Section 5.5**, it is mentioned that the default value of the `t` field is either **implementation-dependent** or **Don't Care**. Why is the default value considered "Don't Care"?
> 

IOPMP doesn't check any transaction when `HWCFG0.enable` = 0. In other worlds,  the `t` fields are ignored when `HWCFG0.enable` = 0. An implementation can be "Don't Care" for the `t` field to save more gate count if the implementation decides that `HWCFG0.enable` is programmable (reset value of `HWCFG0.enable` is 0).

> #### Questions and Observations:
> 1. **Behavior at Reset**:
>    Assuming the MDCFG table is initialized to zero at reset:
>
>    * If software programs MD2 and associates it with an RRID while MD3 [...]

MDCFG Table has improper setting in the scenario, the MDCFG Table enforces implementation-dependent behavior.

>    My understanding is that as long as an MD is not associated with any RRID, the condition in Section 3.3.1 does not apply. Is this interpretation correct?

Not correct. When MDCFG Table has any improper setting, the behavior in MDCFG Table is implementation-dependent. In other words, MDCFG Table outputs the indexes of associated IOPMP entries from request of SRCMD Table when IOPMP checks a transaction. The MDCFG Table still receives same input from SRCMD Table but it will output with alternative behavior when it has any improper setting.

> 2. **Clarification on Restrictions**:
>    The specification imposes the following restrictions:
>
>
>
>    * Are these restrictions applicable **only** to memory domains associated with an RRID?

No, these restrictions are applicable only to MDCFG Table, even if all RRIDs don't associate with all memory domains.

>    * If these restrictions apply to both associated and unassociated MDs, then it seems the software must program the MDs sequentially (e.g., MD0 → MD1 → MD2, etc.). This would prevent configuring MDs in any arbitrary order. Is this the intended behavior?
>    

Programing MDs with specific sequentially is the intended behavior.

>    If the default value of `MDCFG(m).t` is set to the maximum supported IOPMP entries, associating or not associating an MD would not pose an issue. However, this could compromise security if `MDCFG` is not programmed before `SRCMD_EN`.
> 

There are two safe timing for programming:
	1. When HWCFG0.enable is 0, IOPMP doesn't check any transaction. So IOPMP can be programmed.
	2. Software uses "Programming Protection Registers" (MDSTALL, MDSTALLH, RRIDSCP) to stall possible transactions. IOPMP doesn't check stalled transactions. So entries for the stalled transactions in MDCFG Table can be programmed.

Other timings could compromise security.

> #### Example Scenario:
> Consider the following example where, at reset, `MDCFG(m).t = 0` for all `0 <= m <= 62`. Software programs the MDs as follows:
>
> * `MDCFG(2).t = 8` (Associated with RRID A).
> * [...]

The behavior of MDCFG Table in the Example Scenario is implementation-dependent because MDCFG Table has an improper setting at `MDCFG(3).t` (it is less than `MDCFG(2).t`). The SRCMD Table in two buggy scenario are still in implementation-dependent state because `MDCFG(3).t = 0` doesn't be resolved. Though `MDCFG(19).t` doesn't violate the "`MDCFG(m+1).t` >= `MDCFG(m).t`" rule.

> This happens because restriction (2) does not says that the entry index in a higher-indexed memory domain should be greater than **all** indexes in lower-indexed memory domains.
> 
> #### Concern:
> This highlights a potential issue:
> 
> * Detecting such misconfigurations during programming is complex, as it requires searching all MDs for duplicate entries.
> * Multiple MDs can be associated with RRIDs in a single `SRCMD_EN` write, further complicating validation.
> 

IOPMP can simply be programmed without detection.  Misconfigurations can be detected only during transaction checking.
For an example based on your Example Scenario and reference behavior 3 in Section 3.3.1 of IOPMP specification V0.6, no entries belongs to MD3 when IOPMP checks a transaction with RRID A or RRID B.
I think that the reference behavior 3 has less efforts in validation than detecting misconfigurations during programming.

As to detecting by software, software can search `MDCFG(n-1).t` to `MDCFG(m+1).t` when new setting is for `MDCFG(n).t` to `MDCFG(m).t`, `n` <= `m`. The MDCFG Table should always be proper before programming.

> I support the solution proposed in **Issue [#54](https://github.com/riscv-non-isa/iopmp-spec/issues/54)** , which suggests runtime detection and error reporting (e.g., **IOPMP Mis-Configured**) in [...]

Generating error for IOPMP misconfiguration can be a custom feature. I think that it is not a required feature. For example, PMP entries in RISC-V ePMP don't generate runtime error for a potential misconfiguration. Software should properly program them.

> #### Proposal for Software Guidelines:
[... cut ...] ification include a **Software Guidelines** section that explicitly outlines the correct programming sequence for IOPMP configuration? [...]

The IOPMP specification has included a Software Guidelines for runtime programming. It is using "Programming Protection Registers" in Chapter 4 by temporarily stopping transactions until all settings are programmed. Therefore, documenting it may be a little bit wordy because there are two similar functions. If a suitable section has found, we could try it.