riscv-non-isa/riscv-iopmp #76: IOPMP spec review

pperesse, 2025-05-16:
As per the internal review, I'm listing here some comments and questions on the current IOPMP spec:

1. Figure (a/b): X attribute is not a standard transaction type on the bus. This seems to be an implementation choice. Throughout the document, the x transaction type is considered as standard while it isn’t. Typical transaction types on buses are simply read and write types.
2. Figure (b) shows the source-enforcement as there are 2 initiators besides the RISC-V harts. Figure (a) is showing a destination-enforcement, but IOPMP is actually performing intermediate checking in between 2 interconnects. There should be another integration example with IOPMP at each resource, final destination. This implementation allows flexible and scalable integration of checkers in the system.
3. Shouldn’t access restriction to the control port be considered ?  Shouldn’t it be considered that error information is sensitive information that should be restricted to some specified RRID ?
4. The IOPMP configuration closest to the WorldGuard checker is the one with SRCMD Table Format 2 and MDCFG Table Format 1 with k=1 (HWCFG0.md_entry_num=0) which translates to no real concept of Memory Domain (MD) as it implements 1 entry per MD and the permission per RRID defined in the SCRMD table for that MD.
As I expressed earlier in TG meetings, this implementation is quite complex while a different mapping of the entry array to allow r/w permissions per RRID would be simpler to convey and implement. 
5. Section 3.4, should we add a name and a description to figure 5 ?

paul-andes, 2025-05-22:
Regarding the 4th issue, we have been working together for about half a year to minimize our divergences. The outcome was presented to the Summit NA last year. For a user/programmer, the only change is to modify one base address when setting an entry. The following example was demonstrated at that Summit.

// a conceptual function to set an entry in your proposal 
```
set_entry(entry entry_id, perm p, region r, cfg c) {
        ENTRY[entry_id].perm = p;
        ENTRY[entry_id].region = r;
        ENTRY[entry_id].cfg = c;
}

```
// in the IOPMP spec v0.6
```
set_entry(entry entry_id, perm p, region r, cfg c) {
        SRCMD[entry_id].perm = p; // use a different base address
        ENTRY[entry_id].region = r;
        ENTRY[entry_id].cfg = c;
}

```
I believe you can understand the difficulty of merging opinions from several individual entities.

tyshyu, 2025-05-22:
The 1. 2. and 5. are great for document. We will update based on your feedback.
I also clarify the 3. below your feedback.

> 1. Figure (a/b): X attribute is not a standard transaction type on the bus. This seems to be an implementation choice. Throughout the document, the x transaction type is considered as standard while it isn’t. Typical transaction types on buses are simply read and write types.

We will use "(X)" to show that X attribute is not a standard transaction type on the bus and the IOPMP specification has an option for X attribute.

> 2. Figure (b) shows the source-enforcement as there are 2 initiators besides the RISC-V harts. Figure (a) is showing a destination-enforcement, but IOPMP is actually performing intermediate checking [...]

We will add new integration example with IOPMP at each resource or final destination for destination-enforcement. The original Figure 1 (a) will be an example integration for flexibility and scalability.

> 3. Shouldn’t access restriction to the control port be considered ?  Shouldn’t it be considered that error information is sensitive information that should be restricted to some specified RRID ?

IOPMP has local lock bits for basic access restriction.

As to the issue about error information, the control port can be restricted by IOPMP like the right-side IOPMP in Figure 1 (a). The IOPMP can restrict control port of all IOPMPs like transactions to other resources.

Some custom unit also can be implemented for restricting control port but it is out of the specification.

> 5. Section 3.4, should we add a name and a description to figure 5 ?

It should. Do you have any idea for the name?

paul-andes, 2025-06-13:
For (5), we picked up a name.

closed 2025-06-21 by paul-andes