riscv-non-isa/riscv-iopmp #91: [IOPMP Reference Model] R/W/X permission check logic

gagachang, 2025-08-11:
Hi @gullahmed1 

We found some issues about R/W/X permission check logic that we want to discuss with you.

The specification says:
```
Every entry can permit read, write, and instruction fetch of a transaction by
its r, w, and x bits, respectively.
```

However, the reference model checks the write transaction request by R&W bits, and instruction fetch transaction request by R&X bits.
These behaviors may violate the specification.

https://github.com/riscv-non-isa/iopmp-spec/blob/main/iopmp_ref_model/src/iopmp_rule_analyzer.c#L129C1-L141C11
```c
    #if (SRCMD_FMT == 0)
        read_allowed    = sps_en ? (iopmpcfg.r & srcmd_r_bit) : iopmpcfg.r;
        write_allowed   = (sps_en ? (iopmpcfg.w & srcmd_w_bit & iopmpcfg.r & srcmd_r_bit) : (iopmpcfg.w & iopmpcfg.r));
        execute_allowed = (sps_en ? (iopmpcfg.x & srcmd_r_bit) : iopmpcfg.x);
    #elif (SRCMD_FMT == 1)
        read_allowed    = iopmpcfg.r;
        write_allowed   = (iopmpcfg.w & iopmpcfg.r);
        execute_allowed = iopmpcfg.x;
    #elif (SRCMD_FMT == 2)
        read_allowed    = iopmpcfg.r || srcmd_perm_r;
        write_allowed   = ((iopmpcfg.w || srcmd_perm_w) & (iopmpcfg.r || srcmd_perm_r));
        execute_allowed = (iopmpcfg.x || srcmd_perm_r);
    #endif
```

We suggest that the generic write transaction request only requires W bit, and instruction fetch transaction request only requires X bit.

For AMO transaction, there are two cases that can be found in specification:
```
1. To grant an AMO transaction permission, entries and/or memory domains must have read
access permission and write access permission.
2. Some AMO implementations of I/O agents are using a non-atomic read-modify-write
sequence which could contain a read access transaction and a write access transaction, not
single AMO transaction.
```
For AMO, the reference model must check the write transaction request by R&W bits as your current implementation.
The reference model can simulate AMO transactions by adding new flag into `iopmp_trans_req_t` which indicates the transaction is either non-AMO or AMO, and checking additional permission bits for AMO.

gullahmed1, 2025-08-11:
Thank you for your feedback @gagachang 

For instruction fetches, there are two cases where the read permission is checked along with the execute permission:

* Case 1: SRCMD_FMT = 0 when SPS is enabled
* Case 2: SRCMD_FMT = 2

In Case 1 (SRCMD_FMT = 0 with SPS enabled), both permissions, from the IOPMP entry and SRCMD_R/W, must be granted to validate the access. Regarding the execute permission with SPS extension, specification (Appendix A3) states:

> The instruction fetch permission on IOPMP/SPS is considered the same as the read permission.

Therefore, in this case, the read permission from SPS is applied for instruction fetches.

In Case 2 (SRCMD_FMT = 2), SRCMD_PERM.R is also used for instruction fetches. As per Section 3.2.3 of the specification:

> SRCMD_PERM(m).r[s] indicates the read permission and instruction fetch permission of RRID s belonging to the memory domain m.

Accordingly, the read permission from SRCMD_PERM is ORed with the execute permission from the IOPMP entry.

For your second case, write transactions, you are correct:

* For non-AMO, only the write permission is required to validate the access.
* For AMO (atomic) transactions, an additional flag is needed to identify the transaction, and both read and write permissions must be checked.

I'll incorporate this change and mention this issue in PR.

gagachang, 2025-08-11:
> Thank you for your feedback [@gagachang](https://github.com/gagachang)
>
> For instruction fetches, there are two cases where the read permission is checked along with the execute permission:
>
> * Case [...]

Yes, I'd like to summarize the permission check logic.
@tyshyu Please correct me if there is any mistake.

When SRCMD_FMT = 0 and SPS extension is implemented:
- For an incoming read transaction, IOPMP must check both `SRCMD_R` and `ENTRY_CFG.r` are granted
- For an incoming write transaction, IOPMP must check both `SRCMD_W` and `ENTRY_CFG.w` are granted
- For an incoming instruction fetch transaction
  - If `HWCFG0.chk_x=1`,  IOPMP must check both `SRCMD_R` and `ENTRY_CFG.x` are granted
  - If `HWCFG0.chk_x=0`,  IOPMP must check both `SRCMD_R` and `ENTRY_CFG.r` are granted

When SRCMD_FMT = 0 and SPS extension is not implemented:
- For an incoming read transaction, IOPMP must check `ENTRY_CFG.r` is granted
- For an incoming write transaction, IOPMP must check `ENTRY_CFG.w` is granted
- For an incoming instruction fetch transaction 
  - If `HWCFG0.chk_x=1`,  IOPMP must check `ENTRY_CFG.x` is granted
  - If `HWCFG0.chk_x=0`,  IOPMP must check `ENTRY_CFG.r` is granted

> In Case 2 (SRCMD_FMT = 2), SRCMD_PERM.R is also used for instruction fetches. As per Section 3.2.3 of the specification:
>
>
> Accordingly, the read permission from SRCMD_PERM is ORed with the execute [...]

When SRCMD_FMT = 2:
- For an incoming read transaction, IOPMP must check either `SRCMD_PERM.r` or `ENTRY_CFG.r` is granted
- For an incoming write transaction, IOPMP must check either `SRCMD_PERM.w` or `ENTRY_CFG.w` is granted
- For an incoming instruction fetch transaction
  - If `HWCFG0.chk_x=1`,  IOPMP must check either `SRCMD_PERM.r` or `ENTRY_CFG.x` is granted
  - If `HWCFG0.chk_x=0`,  IOPMP must check either `SRCMD_PERM.r` or `ENTRY_CFG.r` is granted

> For your second case, write transactions, you are correct:
>
> * For non-AMO, only the write permission is required to validate the access.
> * For AMO (atomic) transactions, an additional flag is needed [...]

Thanks! Please also CC me when the PR is created.

tyshyu, 2025-08-12:
> Yes, I'd like to summarize the permission check logic. [@tyshyu](https://github.com/tyshyu) Please correct me if there is any mistake.
>
> When SRCMD_FMT = 0 and SPS extension is implemented:
>
> * For an [...]

It is correct when `HWCFG0.no_x = 0`  and `HWCFG0.no_w = 0`.

There is additional information:

According to Section 2.6:
```
There are two read-only bits, HWCFG0.no_w and HWCFG0.no_x, used to decide whether the IOPMP
denies any write transaction and any instruction fetch, respectively.
```
An incoming transaction must check `HWCFG0.no_x` and `HWCFG0.no_w` first, though the check can be ignored if `HWCFG0.no_x = 0` and `HWCFG0.no_w = 0`.

And then check corresponding permissions in `SRCMD_PERM` and `ENTRY_CFG` as your comment.

gagachang, 2025-08-20:
https://github.com/riscv-non-isa/iopmp-spec/pull/94 is merged. Close this issue.